kubi48/TransistorTester-source #1: undefined reference to "__eerd_byte_m2560"

AntonioFromBrazil, 2021-09-07:
I am traying to compile your 1.44 version for ATMega2560 using CodeBlocks 20.03 and I am facing erros undefined reference to "__eerd_byte_m2560" in modules adjust.c,  display.c,  probes.c,  and user.c. The erros happens even if I try to compile for ATMega328... Any idea about what would be happen ? Thanks a lot. Antonio

madires, 2021-09-07:
Possibly a problem with the library path for the avr-gcc standard libs in CodeBlocks.

AntonioFromBrazil, 2021-09-07:
Thanks nadires, but everything in CodeBlocks is equal that ones I have used for version 1.43 that compiles ok. The mentioned erros are issued only compiling version 1.44 !!!

madires, 2021-09-07:
Indeed! For example, adjust.c is the same for 1.43m and 1.44m. No changes at all. Makefile too.