kubi48/TransistorTester-source #5: Can UART output provide same information as LCDs?

Nic30, 2023-10-17:
I have a question about configuration without LCD where all output is send trough UART.
ctester-1.50m.pdf describes this configuration as VT100 driver.

If I underestand correctly this functionality is implemented in lcd-routines-serial.c .
Looking at the https://github.com/kubi48/TransistorTester-source/blob/master/trunk/lcd-routines-serial.c#L301 it seems to me that some features are missing. I am trying to figure out if there is only some icon missing or the whole information is missing.

* Does console output contain for example transistor type and pin numbering?
* Is it possible to control menu trough the console? Is there some example of console output? "2.7.7. Communication with PC" section in doc contains something. But I do not know if it is possible for example execute calibration or pwm tool.

kubi48, 2023-10-18:
The trunk sources refer to the "k" version of the software. 

The "m" versions are in the Markus folder in a packed version. The latest version is ComponentTester-1.50m.tgz. 

As far as I know, the serial output is there in the serial.c routine.

 

In the "k" version, the serial output is in the assembler function swuart.S .

With the "k" version there is only a simplified serial output with ASCII characters. 

For example, a resistor is shown as "1-RR-2" instead of the graphic representation on an LCD display.

 

Karl-Heinz

(Author of the k software)

Gesendet: Mittwoch, 18. Oktober 2023 um 00:41 Uhr

kubi48, 2023-10-24:
> @kubi48 Thank you for your response.
> I am still little bit lost in transistor tester versions.
> Have you by any chance seen any version which can run on STM32 and supports 5v+ (to test mosfets properly)?

I don't know of any STM32 version with 5V operating voltage. 

Both software versions are adapted to the 8-bit AVR with 5V operating voltage.

 
 

Gesendet: Montag, 23. Oktober 2023 um 19:11 Uhr

Nic30, 2023-10-24:
OK, thank you for answer.

closed 2023-10-24 by Nic30