kubi48/TransistorTester-source #6: Component Tester TC1

theos10, 2023-12-10:
I have compiled the firmware k 1.50 for TC1, atmage324pa 16Mhz.
The compilation works, but there is no display.
When I press a button, the screen lights up (white), then goes out a few seconds later, until the next press.
I have tested different display libraries, in config_644.h, but none of them work.
I have a 1.8 INCH TFT128X160 screen. 
Thank you for your help.

kubi48, 2023-12-11:
Hello,

 

you use the "m" firmware, so I can not help you very well.

I developed the "k" firmware, but it is not well suited for the TC1 model 

because the IR capabilities are not supported with "k".

I think, that you must use the ILI9163 library for your display.

Please check, that all interface settings are correct for your TC1 tester.

 

Best regards,

Karl-Heinz Kübbeler

Gesendet: Sonntag, 10. Dezember 2023 um 18:36 Uhr

madires, 2024-01-06:
Please see the 'Clones' file in the source archive. There you'll find several hints about the TC-1 and the recommended settings.

Nickduino, 2024-05-31:
> Please see the 'Clones' file in the source archive.

I couldn't locate it.

The only file I found that mentions the clones is this one:
https://www.mikrocontroller.net/attachment/496540/TableClonesEn.pdf

But it seems it hasn't been updated in a while.

If you had to choose one hardware (kit or assembled) & software combo, what would it be?

madires, 2024-05-31:
The Clones file is in the m-firmware's tar.gz archive. Please download the latest m-firmware tar.gz and extract the Clones file. I would build my own tester because most of the Chinese clones come with a different MCU (not ATmega) at the moment. And their modified firmware is poorly ported.

Nickduino, 2024-06-01:
> Clones file is in the m-firmware's tar.gz archive

Got it. that's the text file with the title "Collection of settings for various tester models"

I was hoping to find a list of clones telling me which ones are ok but if they all come with unsupported chips, I guess I'll wait as I don't feel I can source every component etc. myself. Markus and Karl-Heinz don't sell kits, do they?

madires, 2024-06-01:
We don't sell any Transistortesters. Your best bet would be to buy a clone with a DIP ATmega.