kubi48/TransistorTester-source #9: Another question for the author

izyino, 2024-08-22:
Hello... Since ATmega328, Atmega 2560, etc are a kind of old guys, why not upgrade this fantastic project to something new such ESP32 ? I know it would demand a lot of work but I think is much better put effort in the new microprocessor  then in the old ones !!! Thnks

Hubidrei, 2024-08-22:
I suggest to use a RPi pico. So i am working to do so.

kubi48, 2024-08-22:
Hallo...

I would prefer a microcontroller with a 5V operating voltage for a new design.

A newer type from Microchip would be the AVR128DB64. 

At the moment I am still having problems finding an avr-gcc that supports these newer types. 

The avr-gcc version 14.1 may be suitable, but it is not yet available as a package for my Linux Mint.

 

Best regards,

Karl-Heinz

Gesendet: Donnerstag, 22. August 2024 um 16:38 Uhr

izyino, 2024-08-22:
I do not knot why 5V microcontroler is better than 3.3v ones. The lower Vcc power supply tends to be the majority of then. Do you have some reason to preffer 5V instead ? Thanks

kimstik, 2024-08-23:
ATSAMC20E18A seems 5V one which supported well by GCC