kubi48/TransistorTester-source #11: Software Enhancements Ready for Merge - Taken From v1.53m

KenBC1, 2024-10-30:
Hello,

I'm an embedded developer in Calgary, Alberta, Canada.  I found your project after buying an 2578AY-AT with ST7735 color kit and assembling it.  I enjoyed working in your code base.  My work is entirely non-commercial.

I would like to offer what I think are some significant improvements based upon the V1.53m base.  My only condition is keeping the Copyright on the new files that I created intact (and I agree with the License agreement).

Primary enhancements:
(1) New module stateADCvectISR.c and modified ADC.c to support continuous sampling (13.5 ADC clocks/2 ADC S/H period)
     (a) By default, supports Interrupts on ADC, however, also supports "spin-wait" for timing-critical code (such as SmallResistor())
     (b) Adds ADC_INLINE_DELAY() and ADC_INLINE_DELAYus() to support ADC-based timing in addition to wait()
     (c) Utilizes SLEEP_MODE_ADC during conversions for ADC noise reduction and power saving.
(2) Uses less power with combination of SLEEP_MODE_ADC plus disabling power to TIMERS when not in use.
(3) Used "inline" subroutines in cap.c to make code easier to read and analyze (refactored existing code).
(4) Created reusable subroutine from existing SmallCap(), RefCap(), MeasureInductor()
    (a) StartTimer1AnalogComparison() performs the TIMER1 operation (refactored existing code)
(5) Improvements to External Reference operation:
    (a) Firmware embeds the ASCII version of reference values in flash for fine-tuning without recompile (just use intel .hex editor)
    (b) A new option allows the External Reference to be used in place of INTERNAL BANDGAP
    (c) External Reference allows extended range i.e. 1245 mVolt to 2900 mVolt (e.g. I use lm4041c12)
(6) New support routines added for code clarity dealing with ADC_DDR/_PORT, R_DDR, R_PORT operations.
(7) Improved implementation of MilliSleep() - better code handling of timing calculations.

I am just finishing testing and debugging.  If you would like, I can send the changes to you to review.  If you decide, I can leave the code to you to commit or I can be added as a contributor; you can decide.

I look forward to hearing what you think and if/how you would like to proceed.

Ken

madires, 2024-10-30:
Hi Ken!

Any contribution is welcome! Please email me your improvements (for address see README) and I'll check it out (might take a while since I'm quite busy at the moment).

Cheers,
 Markus

KenBC1, 2024-11-10:
Hi Markus:

Thank you for your invite earlier.

I have been hard at work finalizing the code with bug fixes and also verifying that things still work under various config settings.

I have attached my changes in two files:

- the source code to build the proposed version 1.54m
- the difference in source code from where I started 1.53m

There are several changes and improvements. I plan to compile a spreadsheet with the detailed changelog later. I know you are busy, however, I wanted to get the code to you asap so that you had a change for initial impressions and maybe a chance to compile and try the changes yourself. The first thing that you will notice is the new support for pre-compiled headers and parallel make. For example, if you have 12 cores on your linux machine, you can "make -j 12" and you will find the compilation very fast.
Overall, I am happy with my results and I am glad that you gave me this opportunity to contribute to open source.

I look forward to your comments and an eventual release 1.54m when you have the time. If there is anything you need/desire from me, please let me know.

Ken

Sent with [Proton Mail](https://proton.me/mail/home) secure email.

diff -Naur ComponentTester-1.53m/ADC.c ComponentTester-1.54m/ADC.c
--- ComponentTester-1.53m/ADC.c	2024-11-08 16:51:05.000000000 -0700
+++ ComponentTester-1.54m/ADC.c	2024-11-08 09:41:51.384579140 -0700
@@ -2,7 +2,7 @@
  *
  *   ADC functions
  *
- *   (c) 2012-2023 by Markus Reschke
+ *   (c) 2012-2024 by Markus Reschke
  *   based on code from Markus Frejek and Karl-Heinz Kübbeler
  *
  * ************************************************************************ */
@@ -12,171 +12,633 @@
  *  local constants
  */
 
-/* source management */
-#define ADC_C
-
-
 /*
  *  include header files
  */
 
 /* local includes */
-#include "config.h"           /* global configuration */
-#include "common.h"           /* common header file */
-#include "variables.h"        /* global variables */
-#include "functions.h"        /* external functions */
+#include "projectHeaders.h"
 
+/* source management */
+#define ADC_C
 
-/* ************************************************************************
- *   ADC
- * ************************************************************************ */
-
+/* Define the required in-line functions... */
+#define SCALECHANGE32U16U
 
-/*
- *  read ADC channel and return voltage in mV
- *  - use Vcc as reference by default
- *  - switch to bandgap reference for low voltages (< 1.0V) to improve
- *    ADC resolution
- *  - with a 125kHz ADC clock a single conversion needs about 0.1ms
- *    with 25 samples we end up with about 2.6ms
- *
- *  requires:
[... 1597 lines cut ...]
+    {
+        goto BAILOUT_RETURN;    /* Major issue with firmware. */
+    }
+
+    Cfg.Vcc     = VccRefNominalValue;   /* (re)Start from nominal value. */
+
+    /* First call to ChangeADCRef() will delay when required. */
+    Channel     = ADC_REF_VCC | TP_REF; /* Force VCC reference */
+
+    /* Note: Interrupts are allowed here if enabled.
+    **  This is used because the ReadUSingleSample() routine will not force
+    **          interrupts on; this is the way to make sure ADC ints are on.
+    */
+    ReadU(Channel);     /* Dummy Read */
+
+#if defined(HW_EXTREF)
+    /*
+    ** External voltage reference is expected - get the specified value.
+    */
+    if (   !(MyURef = ParseFlashU32(&ADCextReference[0], &ExtRefNominalValue)))
+    {
+        goto BAILOUT_RETURN;    /* Major issue with firmware. */
+    }
+
+    /* For high accuracy make reference measurements ove

madires, 2024-11-11:
Hi Ken!

Looks like GitHub added one attachement to the message, truncated it around 64kB, and dropped both file attachements. I guess, GitHub doesn't support that via email. Please try  the 'Attach files' button of the comment editor or send a direct email.  

Cheers,
 Markus