kubi48/TransistorTester-source #12: V1.53m Super-Beta BJT Base-Emitter Reverse Breakdown is Possible

KenBC1, 2024-11-17:
I happened to use a high-gain, small signal BJT (2N5088) in my testing and observed intermittent issues with the Tester's reported hFE.  Most of the time, the Tester responds with very reasonable hFE for my transistor as follows:

NPN
-----------
hFE  520 c
I_E    2.742 mA
Vbe   618 mV

Fortunately, my testing repeated this measurement several times.  After a while, the Tester would respond with much too high hFE for this transistor as follows:

NPN
-----------
hFE  6123 c
I_E    6.671 mA
Vbe   621 mV

I looked at the code closely and observed that the Tester is applying Vcc (5 volts) with Rl (680 ohms) to the collector-base (C-B) junction (this is normal and okay).  There are times where the Tester applies this signal to the emitter-base instead; this can be a problem with Super-Beta BJT's because the reverse breakdown potential is low (e.g. for 2N5088 breakdown is 4.5 volts).  In my case, it seems that my transistor broke-down after many tests.

I have read that the forward gain of Super-Beta BJT's will degrade once the transistor has been operated in reverse emitter-base breakdown.  Ideally, the Tester should only apply Vcc/2 (i.e. about 2.5 volts) to C-B and E-B junctions until the Tester determines the proper orientation and then it can use full Vcc (5 volts) on the C-B junction.

An alternate solution might be to use Rh (470 k ohms) on the C-B and E-B junctions and then switch to Rl (680 ohms) when the proper orientation is found?  The emitter-base junction may still break down in this case, however, the low resulting current might save the transistor some stress.

I'm continuing to play with this and I might be able to work-in some changes for release V1.54m?  Until then, I just wanted people to know of this potential issue.

madires, 2024-11-17:
Hi Ken!

Thanks for sharing your findings! Did the pinout change too (emitter and collector)? If yes, which one was the correct pinout (for the normal beta or the high beta)? There are some points which don't make much sense to me at the moment. The maximum possible currect in case of a reverse emitter-base breakdown at 4.5 V would be just 0.7 mA ( (5 V - 4.5 V) / (680 Ohms + 20 Ohms + 20 Ohms), 20 Ohms is the IO pin's internal resistance). Not much current to damage a small signal BJT. If really damaged I would expect a short (Vbe is 621 mV). Also, if repeated reverse emitter-base breakdowns degrade hFE, the value should decrease and not be way too high. Looks to be something different.

Cheers,
 Markus

KenBC1, 2024-11-17:
Thanks Markus.

I agree, I need to do more investigation.  When my high-beta started failing then the Tester would show 6 mA of collector current instead of the usual 2... mA.  When I kept trying, the high-beta transistor completely failed; after this the Tester started thinking it was a low-current N-channel depletion FET!

Not to worry yet - I have another high-beta transistor and I plan to dig in deeper.

It could be something different.  You are correct, the reverse breakdown should be zener-like behavior but what I saw seemed to indicate that the collector-emitter junction went to zero instead of ~4.5 volts; in that case the current might be expected to go as high as 5V/720ohms = 6.9 mA.  All I know for sure is that the tester showed > 6 mA for I_c in this strange condition.

I have been adding good debug information temporarily to the code so that I know more information when/if it happens again.  On a side note, it sure is nice to have a 8-line color display for this debugging.  :-)

Ken