kubi48/TransistorTester-source #13: feature request >> ADD SHORT CIRCUIT TESTER WITH BUZZER

clear-sky, 2024-12-08:
Hi 
if possible please add short circuit tester with buzzer if possible! like most multimeters that beeps if connection resistance is lower than 50 ohms

clear-sky, 2024-12-08:
in this path i found something:
compenentTester/configMCU.h for example config_328.h
please see:
```
/*
 *  buzzer
 */

#define BUZZER_PORT      PORTD     /* port data register */
#define BUZZER_DDR       DDRD      /* port data direction register */
#define BUZZER_CTRL      PD5       /* control pin (low: off / high: on) */
```
but could not find any documentation about schematic and its usage

clear-sky, 2024-12-08:
I have decided to make this project with m328 and want to enable IR reciever and HW buzzer enable but in config_328.h saw that for both IR and buzzer only PD5 has been defined!
how can i do wiring?

madires, 2024-12-08:
> Hi if possible please add short circuit tester with buzzer if possible! like most multimeters that beeps if connection resistance is lower than 50 ohms

There's already a continuity check (SW_CONTINUITY_CHECK) which uses the buzzer. Anyhow, I would recommend to use a good DMM which should have a lower response time.

madires, 2024-12-08:
> in this path i found something: compenentTester/configMCU.h for example config_328.h please see:
>
> ```
> /*
>  *  buzzer
>  */
>
> #define BUZZER_PORT      PORTD     /* port data register */
> #define [...]

To enable the buzzer option and to select the type edit config.h:
/*
 *  Buzzer
 *  - see BUZZER_CTRL in config_<MCU>.h for port pin
 *  - buzzer types
 *    BUZZER_ACTIVE: active buzzer with integrated oscillator
 *    BUZZER_PASSIVE: passive buzzer
 *  - uncomment to enable and also select the correct buzzer type
 */

//#define HW_BUZZER
#define BUZZER_ACTIVE                   /* active buzzer */
//#define BUZZER_PASSIVE                  /* passive buzzer */

> but could not find any documentation about schematic and its usage

From the README:
+ Buzzer (hardware option)

If you like to have an audible indicator you can add an active buzzer (with
integrated oscillator) or a passive buzzer. The buzzer is controlled via a
fixed pin (see BUZZER_CTRL in config_<MCU>.h) and can be driven directly (
< 20 mA) or by a transistor (NPN or n-channel MOSFET).

The passive buzzer generates short beeps with 2.5 and 5 kHz.

madires, 2024-12-08:
> I have decided to make this project with m328 and want to enable IR reciever and HW buzzer enable but in config_328.h saw that for both IR and buzzer only PD5 has been defined! how can i do wiring?

Choose an unused IO pin and change BUZZER_CTRL (also BUZZER_PORT and BUZZER_DDR) accordingly. BTW, you can select SW_IR_RECEIVER to use the probes for connecting the IR receiver module. A dedicated IO pin is only needed for HW_IR_RECEIVER.

clear-sky, 2024-12-08:
Hi -Thanks for your detailed answer
many many thanks

closed 2024-12-08 by clear-sky