kubi48/TransistorTester-source #14: BAT_POOR variable , used in  /Battery_check.c  where is it denfined before ?

leoengcea1, 2025-01-03:
in this file https://github.com/kubi48/TransistorTester-source/blob/master/trunk/Battery_check.c 
BAT_POOR variable is useb for calculations but is not defined before anywhere . where is it defined ? is it other file what file ?

kubi48, 2025-01-04:
Hello Leonardo,
the BAT_POOR value is defined in the Makefile like many other constants.
Please read any Makefile you can find in subdirectories of the directory trunk.
The idea of ​​this construction is not to have to change any source code for the different tester versions.
Any special setting is done in the Makefile, which is different for any tester version.
You can specify a new subdirectory of the directory trunk and can copy a Makefile from
a directory with a simular configuration and change this Makefile.
For example you can change the operating language (UI_LANGUAGE) or the poor level of the supply battery.
You can create the new program by running the make program in your subdirectory.
 
Best regards.
Karl-Heinz
 

Gesendet: Freitag, 3. Januar 2025 um 18:25

nvdl, 2026-11-10:
Can this issue be closed?